adafruit/Adafruit_CircuitPython_ESP_ATcontrol #4: JSON errors

jerryneedell, 2018-12-13:
FYI - I've seen this a few times
```
<--- b'\r\nRecv 66 bytes\r\n\r\nSEND OK\r\n'
[b'HTTP/1.1 200 OK', b'Content-Type: application/javascript', b'Content-Length: 673', b'Connection: keep-alive', b'Access-Control-Allow-Origin: *', b'Cache-Control: max-age=15', b'Date: Wed, 12 Dec 2018 22:45:07 GMT', b'Expires: Wed, 12 Dec 2018 22:46:07 UTC', b'Server: nginx/1.12.1', b'X-Powered-By: Fat-Free Framework', b'Age: 8', b'X-Cache: Hit from cloudfront', b'Via: 1.1 77aa002baa7dabd52aea1d477a796cac.cloudfront.net (CloudFront)', b'X-Amz-Cf-Id: 6q6R_9yS7n0G3OZTw2WnoV8sd29UBkMvJMN2rsDc7q7hzh0Hj63Eiw==', b'', b'']
---> AT+CIPCLOSE
<--- b'ec 12, 2018 at 22:45 GMT"},"disclaimer":"i a a rue r h iDsBtondsnhulcneinaefmoexhnrtsr""ata""ion,p"{S"{oe"S"smo:&3""t""4893,ecpin"ne ae lr,refa"4853,B""o""P,sbl"pn;"a""7742,ecpo"Bis udtrn"rtfo"2782,U""o""R,yb"&u;,ae",1963","description":"Euro","rate_float":3041.9963}}}CLOSED\r\n\r\nOK\r\n'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "bc.py", line 47, in <module>
  File "bc.py", line 40, in <module>
ValueError: syntax error in JSON
>>> 
```
Probably just a garbled message and we just need to catch the exception.
Have you run into it?

ladyada, 2018-12-13:
oosps yeah this is my current loop
```
while True:
    try:
        display.print('----')
        # Connect to WiFi if not already
        print("Connected to", esp.remote_AP)
        if esp.remote_AP[0] != MY_SSID:
            esp.join_AP(MY_SSID, MY_PASS)
            print("My IP Address:", esp.local_ip)
        # great, lets get the JSON data
        print("Retrieving price...", end='')
        header, body = esp.request_url(URL)
        print("OK")
    except RuntimeError as e:
        print("Failed to connect, retrying")
        print(e)
        continue

    try:
        print("Parsing JSON response...", end='')
        json = ujson.loads(body)
        bitcoin = json["bpi"]["USD"]["rate_float"]
        print("USD per bitcoin:", bitcoin)
        display.print(int(bitcoin))
    except ValueError:
        print("Failed to parse json, retrying")
        continue

    gc.collect()
    print("Free memory:", gc.mem_free() / 1024)
    time.sleep(15)
```

jerryneedell, 2018-12-14:
fix with try/except

closed 2018-12-14 by jerryneedell