adafruit/Adafruit_CircuitPython_ESP_ATcontrol #5: AT_RST causes nRF52840 UART to hang

jerryneedell, 2018-12-14:
I'v noticed that if an AT+RST command is executed na nrf52840 board communication with the uart appears to lock up.  

This is done routinely in the espatcommand_simpltest.py by the lines
```
if not esp.soft_reset():
    esp.hard_reset()
    esp.soft_reset()
```

it works fine if I just comment these out for now, but a real fix should be found but anyone using an nrf52840 should be aware of this.

This does not happen on the SAMD51...
AT+RST does generate a big blast of data across the UART so there may be something in the nrf52 that does not handle this well.

ladyada, 2018-12-14:
yeah its a burst - if you can make it reproducible we could file an issue !

jerryneedell, 2018-12-14:
yup -- If I can't find a way to clear it,  I'll see if I can come up with a good way to trigger it without an AT ESP8266 as the trigger then file an issue.  It'll be hard for anyone to debug if you need an AT ESP8266 ;-)

jerryneedell, 2018-12-17:
One way to clear this condition is to deinit then re-init the uart.  Would it be acceptable to pass the TX/RX pins into espatcommands and let it create the uart so it can also it can deinit/init the uart if necessary?

ladyada, 2018-12-17:
i want to avoid passing in pins cause then you can't use pyserial. you could always call deinit() whenever you like tho, want to do that before/after hardreset?
also, its a CP issue, @hathach will be looking at it in about a week (he needs to get hardware)

jerryneedell, 2018-12-17:
opened https://github.com/adafruit/circuitpython/issues/1407  
if we deinit, we can't re-init since we won't know pins used... unless there is a way to find them.  I think we should just fix it in CP.

ladyada, 2018-12-17:
oh yeah hmm good point

hathach, 2018-12-26:
I just received the Huzzarh hardware, is setting up circuitpython port to test with.

ladyada, 2018-12-28:
this is... fixed?

closed 2018-12-28 by ladyada