adafruit/Adafruit_CircuitPython_ESP_ATcontrol #12: occasional error in "cheerlights"

jerryneedell, 2018-12-27:
The cheerlights demo occasionally fails with this error,
probably just need to check for non zero length response. I happens after several hours of running.
This was on a metrom4 with esp32 but I have also seen it on the particle_argon
```
---------------------------------------
Parsing JSON response...parsed OK!
179 None value: #FFFF00
132080
---> ATE0
<--- b'\r\nOK\r\n'
---> AT+CIPSTATUS
<--- b'STATUS:4\r\n\r\nOK\r\n'
Retrieving data source...---> AT+CIPSTATUS
<--- b'STATUS:4\r\n\r\nOK\r\n'
---> AT+CIPSTART="TCP","api.thingspeak.com",80,10
<--- b'CONNECT\r\n\r\nOK\r\n'
---> AT+CIPSTATUS
<--- b'STATUS:3\r\n+CIPSTATUS:0,"TCP","34.238.111.95",80,14020,0\r\n\r\nOK\r\n'
---> AT+CIPSEND=109
<--- b'\r\nOK\r\n'
<--- b''
Receiving: 1278
[b'HTTP/1.1 200 OK', b'Content-Type: application/json; charset=utf-8', b'Transfer-Encoding: chunked', b'Connection: close', b'Status: 200 OK', b'X-Frame-Options: SAMEORIGIN', b'Access-Control-Allow-Origin: *', b'Access-Control-Allow-Methods: GET, POST, PUT, OPTIONS, DELETE, PATCH', b'Access-Control-Allow-Headers: origin, content-type, X-Requested-With', b'Access-Control-Max-Age: 1800', b'ETag: "fe85a57b0ed73af20b3f26ad16fcc17a"', b'Cache-Control: max-age=0, private, must-revalidate', b'X-Request-Id: 688ca56c-fb36-4edb-8958-5d9fe85ea4a4', b'X-Runtime: 0.017284', b'X-Powered-By: Phusion Passenger 4.0.57', b'Date: Wed, 26 Dec 2018 03:30:35 GMT', b'Server: nginx/1.9.3 + Phusion Passenger 4.0.57', b'', b'266', b'{"channel":{"id":1417,"name":"CheerLights","description":"CheerLights is an \xe2\x80\x9cInternet of Things\xe2\x80\x9d project created by Hans Scharler that allows people\xe2\x80\x99s lights all across the world to synchronize to one color set by Twitter. This is a way to connect physical things with social networking experiences.","latitude":"40.5","longitude":"-80.22","field1":"Last CheerLights Command","field2":"CheerLights HEX Color","created_at":"2011-12-01T03:30:04Z","updated_at":"2018-07-20T01:05:15Z","last_entry_id":498410},"feeds":[{"created_at":"2018-12-26T03:23:27Z","entry_id":498410,"field1":"yellow","field2":"#FFFF00"}]}', b'0', b'', b'']
---> AT+CIPCLOSE
<--- b'\r\nERROR\r\n'
Reply is OK!
---------------------------------------- Size:  626
266
{"channel":{"id":1417,"name":"CheerLights","description":"CheerLights is an “Internet of Things” project created by Hans Scharler that allows people’s lights all across the world to synchronize to one color set by Twitter. This is a way to connect physical things with social networking experiences.","latitude":"40.5","longitude":"-80.22","field1":"Last CheerLights Command","field2":"CheerLights HEX Color","created_at":"2011-12-01T03:30:04Z","updated_at":"2018-07-20T01:05:15Z","last_entry_id":498410},"feeds":[{"created_at":"2018-12-26T03:23:27Z","entry_id":498410,"field1":"yellow","field2":"#FFFF00"}]}
0

----------------------------------------
Parsing JSON response...parsed OK!
180 None value: #FFFF00
132080
---> ATE0
<--- b'\r\nOK\r\n'
---> AT+CIPSTATUS
<--- b'STATUS:4\r\n\r\nOK\r\n'
Retrieving data source...---> AT+CIPSTATUS
<--- b''
---> AT+CIPSTATUS
<--- b'\r\nERROR\r\n'
---> AT+CIPSTATUS
<--- b'STATUS:4\r\n\r\nOK\r\n'
---> AT+CIPSTART="TCP","api.thingspeak.com",80,10
<--- b'CONNECT\r\n\r\nOK\r\n'
---> AT+CIPSTATUS
<--- b'STATUS:3\r\n+CIPSTATUS:0,"TCP","34.238.111.95",80,48772,0\r\n\r\nOK\r\n'
---> AT+CIPSEND=109
<--- b'\r\nOK\r\n'
<--- b''
[b'']
---> AT+CIPCLOSE
<--- b'CLOSED\r\n\r\nOK\r\n'
Reply is OK!
---------------------------------------- Size:  0

----------------------------------------
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "metrom4_cheerlights.py", line 86, in <module>
IndexError: list index out of range
>>> 

```

ladyada, 2018-12-27:
huh odd, wanna add a try/except there?

jerryneedell, 2018-12-27:
sure -- I'll test it and put a PR in after trigger it.

PR #13 (2018-12-27): add try/except to cheerlights demo
fixes #12 

```<--- b'\r\nOK\r\n'
<--- b''
[b'']
---> AT+CIPCLOSE
<--- b'CLOSED\r\n\r\nOK\r\n'
Reply is OK!
---------------------------------------- Size:  0

----------------------------------------
IndexError processing Body, retrying
 list index out of range
---> ATE0
<--- b'\r\nOK\r\n'
---> AT+CIPSTATUS
<--- b'STATUS:4\r\n\r\nOK\r\n'
Retrieving data source...---> AT+CIPSTATUS
<--- b'STATUS:4\r\n\r\nOK\r\n'
---> AT+CIPSTART="TCP","api.thingspeak.com",80,10
<--- b'CONNECT\r\n\r\nOK\r\n'
---> AT+CIPSTATUS
<--- b'STATUS:3\r\n+CIPSTATUS:0,"TCP","52.0.246.0",80 [... cut]
closed 2018-12-27 by ladyada