adafruit/Adafruit_CircuitPython_ESP_ATcontrol #30: simpletest and webclientseem to be working on the Particle Argon

Arudinne, 2019-05-26:
Hello,

I was waiting for CircuitPython 4.x to be released so that I could try it out with the Argons and Xenons I have. now that it's out I've got 4.0.1 installed on one of my Argons and I can't seem to get either of the Wi-Fi tests to work.

When I run either test it just loops over and over with the following text:

Failed to connect, retrying
 Didn't get IP address
Connected to None
no IP

=====

The simpletest does list the wifi networks I have nearby, but fails past that point:
[image]

=====

I enabled debug mode for the webclient test and it seems to connect (it says WIFI connected at least) but some reason it isn't getting an IP. This happens with either of my WiFi APs
[image]
[image]

These APs are different brands and use different channels (11 and 1). The main SSID "Destiny Ascension" previously worked with the Argons while using the particle firmware.

=====

Am I doing something wrong?

Any help getting this working would be greatly appreciated.

jerryneedell, 2019-05-26:
FWIW - I just updated an Argon to the latest master ( 4.0.1 with a few updates) and it works for me with the esp32at_control library.

Arudinne, 2019-05-27:
I tried the files from the the master branch as well as the files from the release page.

I just now deleted the files under **lib\adafruit_espatcontrol** and installed the ones from **adafruit-circuitpython-espatcontrol-4.x-mpy-0.3.1.zip**. I am also have all of the rest of the files from **adafruit-circuitpython-bundle-4.x-mpy-20190525.zip** under lib,

I have also tried reinstalling the ESP32 firmware from this page twice just in case I somehow flashed it incorrectly: **https://learn.adafruit.com/adding-a-wifi-co-processor-to-circuitpython-esp8266-esp32/program-particle-argon**.

I am running the examples from **adafruit-circuitpython-espatcontrol-examples-0.3.1.zip** and the ONLY change I have made to the test code is to uncomment the lines for the argon and remove the lines for the M4:

`import time
import board
import busio
from digitalio import DigitalInOut
from adafruit_espatcontrol import adafruit_espatcontrol

# Get wifi details and more from a secrets.py file
try:
    from secrets import secrets
except ImportError:
    print("WiFi secrets are kept in secrets.py, please add them there!")
    raise

# With a Particle Argon

RX = board.ESP_TX
TX = board.ESP_RX
resetpin = DigitalInOut(board.ESP_WIFI_EN)
rtspin = DigitalInOut(board.ESP_CTS)
uart = busio.UART(TX, RX, timeout=0.1)
esp_boot = DigitalInOut(board.ESP_BOOT_MODE)
from digitalio import Direction
esp_boot.direction = Direction.OUTPUT
esp_boot.value = True

print("ESP AT commands")
esp = adafruit_espatcontrol.ESP_ATcontrol(uart, 115200,
                                          reset_pin=resetpin, rts_pin=rtspin, debug=False)
print("Resetting ESP module")
esp.hard_reset()

first_pass = True
while True:
    try:
        if first_pass :
            print("Scanning for AP's")
            for ap in esp.scan_APs():
                print(ap)
            print("Checking connection...")
            # secrets dictionary must contain 'ssid' and 'password' at a minimum
            print("Connecting...")
            esp.connect(secrets)
            print("Connected to AT software version ", esp.version)
            print("IP address ", esp.local_ip)
            first_pass = False
        print("Pinging 8.8.8.8...", end="")
        print(esp.ping("8.8.8.8"))
        time.sleep(10)
    except (ValueError,RuntimeError, adafruit_espatcontrol.OKError) as e:
        print("Failed to get data, retrying\n", e)
        print("Resetting ESP module")
        esp.hard_reset()
        continue
`

This is the output I get:

`Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.1 on 2019-05-22; Particle Argon with nRF52840
>>> import esp_atcontrol_simpletest.py
ESP AT commands
Resetting ESP module
Scanning for AP's
[3, 'Destiny Ascension', -37, 'e4:f4:c6:08:e0:bd', 11]
[3, 'Destiny Ascension - Guest', -53, 'ea:33:81:0f:f5:61', 1]
[3, 'ATT2TxFxi2', -64, '88:96:4e:21:22:50', 1]
[3, 'WIFIDC114B', -74, '54:13:79:dc:11:4f', 11]
[0, 'ESP_F0B2BD', -84, '5e:cf:7f:f0:b2:bd', 11]
Checking connection...
Connecting...
Connected to None
no IP
Failed to connect, retrying
 Didn't get IP address
Connected to None
no IP
Failed to connect, retrying
 Didn't get IP address
Connected to None
no IP
Failed to connect, retrying
 Didn't get IP address
Connected to None
no IP
Failed to connect, retrying
 Didn't get IP address
`

I have tried the settings for both of my APs in the secrets files and they aren't working with the Argon but I know my APs are working fine because I have a pyportal that connects to my main WiFi network along with an ESP8266 and several other devices.  An XBOX One connects to my guest network without issues.

I have tried running storage.erase_filesystem() and reinstalling all of the libraries and examples several times, but the tests still fail. I have even tried reflashing the circuity python firmware - **adafruit-circuitpython-particle_argon-en_US-4.0.1.uf2** to no effect.

jerryneedell, 2019-05-27:
I'm baffled - I copied your code (note, I had to insert "import time" at the top -- I think it just got cut off your post)  It runs for me.

For lack of any better ideas, have you tried resetting your AP? I have found that there can be some confusing "caching" of information between the ESP32 and AP ... just guessing...

Arudinne, 2019-05-27:
Just now I tried resetting the main AP and it had no effect. I also tried the lastest version they have on the S3 repo but it also had no effect.

```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.1-2-g63b253c on 2019-05-22; Particle Argon with nRF52840
>>> import esp_atcontrol_simpletest
ESP AT commands
Resetting ESP module
Scanning for AP's
[3, 'Destiny Ascension', -43, 'e4:f4:c6:08:e0:bd', 11]
[3, 'Destiny Ascension - Guest', -53, 'ea:33:81:0f:f5:61', 1]
[3, 'ATT2TxFxi2', -65, '88:96:4e:21:22:50', 1]
[3, 'MySpectrumWiFi80-2G', -78, '94:91:7f:74:65:7e', 5]
[3, 'WIFIDC114B', -84, '54:13:79:dc:11:4f', 11]
[3, 'MySpectrumWiFiaa-2G', -88, '84:a1:d1:8f:f8:b0', 11]
[3, 'ATTPSAYD7i', -89, '74:8a:0d:75:aa:70', 11]
[0, 'ESP_F0B2BD', -91, '5e:cf:7f:f0:b2:bd', 11]
Checking connection...
Connecting...
Connected to None
no IP
Failed to connect, retrying
 Didn't get IP address
Connected to None
no IP
Failed to connect, retrying
 Didn't get IP address
Connected to None
no IP
Failed to connect, retrying
 Didn't get IP address
Connected to None
no IP
Failed to connect, retrying
 Didn't get IP address
Connected to None
no IP
Failed to connect, retrying
 Didn't get IP address
Connected to None
```

I'd like to try the latest master of Circuit Python to see if that's the issue. Unfortunately I'm not really setup for compiling circuitpython myself... the last time I tried to compile 3.x for the NRF52832 Feather I ran into one failure after another.

Would you be willing or able to share your uf2 for 4.0.1-5-g63b253c33 so I can see if that works on mine?

jerryneedell, 2019-05-27:
I can but I think it is the same as the S3 build -- It has the same commit hash -- just a different date since I build it on 5/24

jerryneedell, 2019-05-27:
ah - here it is -- just had to zip it
[adafruit-circuitpython-particle_argon-en_US-20190524-63b253c.zip](https://github.com/adafruit/Adafruit_CircuitPython_ESP_ATcontrol/files/3224888/adafruit-circuitpython-particle_argon-en_US-20190524-63b253c.zip)

Arudinne, 2019-05-27:
Thanks for the upload. Unfortunately it still fails on the Argon I've been testing with. I have another laying around that I'll try testing with later but both worked fine using the Particle firmware so I have doubts that it's a hardware issue.

Are you running the same ESP32 firmware from this page? https://learn.adafruit.com/adding-a-wifi-co-processor-to-circuitpython-esp8266-esp32/firmware-files

jerryneedell, 2019-05-28:
Yes, it is the same version.
One question. You do have the WIFi antenna connected to the Argon, correct?
I expect you do since you see the APIDs but just want to be sure.

Arudinne, 2019-05-28:
Yes, I have a WiFi antenna connected to both argon boards and they're in the WiFi port not the BT port. My desk is within 10 feet of both access points, so range shouldn't be an issue.

This makes absolutely no sense to me. I'll have to try the second argon later today and see if it fares any better.

Arudinne, 2019-05-29:
Both of my argons have this issue, so I'm not sure what's up. I'll have to see if I can test with a different AP at work or something.

jerryneedell, 2019-05-29:
Trying another AP will be very instructive.  Good luck!

jerryneedell, 2019-05-29:
It will be interesting to see if a different AP works. 
Looking at the code, it is failing here:
https://github.com/adafruit/Adafruit_CircuitPython_ESP_ATcontrol/blob/master/adafruit_espatcontrol/adafruit_espatcontrol.py#L441
No idea why but the parsing to the "at_response" has always been s source of problems with this library.

I will try to do some testing on mine to see how it's responses are different from yours.

jerryneedell, 2019-05-29:
Here is my system  debug response to the webclient test 
Note the "GOT IP" response as expected -- Yours did not include this
```
>>>import webclient
ESP AT commands
ESP AT GET URL http://wifitest.adafruit.com/testwifi/index.html
Resetting ESP module
Checking connection...
---> AT
<--- b'AT\r\n\r\nOK\r\n'
---> ATE0
<--- b'ATE0\r\n\r\nOK\r\n'
Changing baudrate to: 115200
---> AT+UART_CUR=115200,8,1,0,2

---> AT
<--- b'\r\nOK\r\n'
---> AT+GMR
<--- b'AT version:1.3.0.0-dev(6aafec8 - Dec  7 2018 02:23:19)\r\nSDK version:v3.3-dev-343-g7fa98593b\r\ncompile time:Dec 24 2018 12:27:05\r\n\r\nOK\r\n'
---> AT+CIPMUX?
<--- b'+CIPMUX:0\r\n\r\nOK\r\n'
---> AT+CIPSSLSIZE=4096
<--- b'ERR CODE:'
---> AT+CIPSSLCCONF?
<--- b'+CIPSSLCCONF:0,0,0,0\r\n+CIPSSLCCONF:1,0,0,0\r\n+CIPSSLCCONF:2,0,0,0\r\n+CIPSSLCCONF:3,0,0,0\r\n+CIPSSLCCONF:4,0,0,0\r\n\r\nOK\r\n'
---> ATE0
<--- b'\r\nOK\r\n'
Changing baudrate to: 115200
---> AT+UART_CUR=115200,8,1,0,2

---> AT
<--- b'\r\nOK\r\n'
---> AT+CIPSTATUS
<--- b'STATUS:0\r\n\r\nOK\r\n'
Connecting...
---> AT+CIPSTATUS
<--- b'STATUS:0\r\n\r\nOK\r\n'
Connected to None
---> AT+CWMODE?
<--- b'+CWMODE:1\r\n\r\nOK\r\n'
---> AT+CIPSTATUS
<--- b'STATUS:0\r\n\r\nOK\r\n'
---> AT+CWJAP="Needell Airport","REDACTED"
<--- b'WIFI CONNECTED\r\nWIFI GOT IP\r\n'
---> AT+CIPSNTPCFG=1,-5
<--- b'ERR CODE:'
---> AT+CIPSNTPCFG=1,-5
<--- b'\r\nOK\r\n'
---> AT+CIFSR
<--- b'+CIFSR:STAIP,"10.0.0.25"\r\n+CIFSR:STAMAC,"30:ae:a4:d1:69:6c"\r\n\r\nOK\r\n'
My IP Address: 10.0.0.25
---> ATE0
<--- b'\r\nOK\r\n'
Changing baudrate to: 115200
---> AT+UART_CUR=115200,8,1,0,2

---> AT
<--- b'\r\nOK\r\n'
---> AT+CIPSTATUS
<--- b'STATUS:2\r\n\r\nOK\r\n'
Retrieving URL...---> AT+CIPDOMAIN="wifitest.adafruit.com"
<--- b'+CIPDOMAIN:104.236.193.178\r\n\r\nOK\r\n'
---> AT+CIPSTATUS
<--- b'STATUS:2\r\n\r\nOK\r\n'
---> AT+CIPSTART="TCP","104.236.193.178",80,10
<--- b'CONNECT\r\n\r\nOK\r\n'
---> AT+CIPSTATUS
<--- b'STATUS:3\r\n+CIPSTATUS:0,"TCP","104.236.193.178",80,50703,0\r\n\r\nOK\r\n'
---> AT+CIPSEND=35
<--- b'\r\nOK\r\n'
<--- b''
---> AT+CIPSEND=29
<--- b'\r\nOK\r\n'
<--- b''
---> AT+CIPSEND=36
<--- b'\r\nOK\r\n'
<--- b''
---> AT+CIPSEND=2
<--- b'\r\nOK\r\n'
<--- b''
Receiving: 310
Status: 200
Content type: text/html
Content size: 70
Encoding: utf-8
---> AT+CIPCLOSE
<--- b'\r\nERROR\r\n'
Text: This is a test of Adafruit WiFi!
If you can read this, its working :)

```

Arudinne, 2019-05-29:
Mine never gets an IP as far as I can tell and I'm not sure why. I've connected Feather ESP8266s and Feather ESP32s without issue to my home network without issue.

I just now tried it on out guest network at work and it gives me the same behavior. 

Just to be sure everything was clean I ran storage.erase_filesystem(), downloaded **Adafruit_CircuitPython_ESP_ATcontrol-master.zip** and copied over the libraries and simpletest example. The only file I kept from the Argon was secrets.py which I edited to match our guest network's credentials. There are no other library files on this argon.

My company's Guest network has no captive portal and uses normal WPA2 encryption, so it's just a password we change on a regular interval. My phone is connecting to it just fine.

-----

Some further testing:

I cut and paste the code you provided earlier and I get the same issue.

I've also tried a slightly older version of libraries from **adafruit-circuitpython-espatcontrol-4.x-mpy-0.3.0** with no change.

I had to redact the output but as you can see it's basically the same:

'''''
Adafruit CircuitPython 4.0.1-5-g63b253c33 on 2019-05-24; Particle Argon with nRF52840
>>> import esp_atcontrol_simpletest
ESP AT commands
Resetting ESP module
Scanning for AP's
---> AT
<--- b'AT\r\n\r\nOK\r\n'
---> ATE0
<--- b'ATE0\r\n\r\nOK\r\n'
Changing baudrate to: 115200
---> AT+UART_CUR=115200,8,1,0,2

---> AT
<--- b'\r\nOK\r\n'
---> AT+GMR
<--- b'AT version:1.3.0.0-dev(6aafec8 - Dec  7 2018 02:23:19)\r\nSDK version:v3.3-dev-343-g7fa98593b\r\ncompile time:Dec 24 2018 12:27:05\r\n\r\nOK\r\n'
---> AT+CIPMUX?
<--- b'+CIPMUX:0\r\n\r\nOK\r\n'
---> AT+CIPSSLSIZE=4096
<--- b'ERR CODE:'
---> AT+CIPSSLCCONF?
<--- b'+CIPSSLCCONF:0,0,0,0\r\n+CIPSSLCCONF:1,0,0,0\r\n+CIPSSLCCONF:2,0,0,0\r\n+CIPSSLCCONF:3,0,0,0\r\n+CIPSSLCCONF:4,0,0,0\r\n\r\nOK\r\n'
---> AT+CWMODE?
<--- b'+CWMODE:1\r\n\r\nOK\r\n'
---> AT+CWLAP
<REDACTED>
[3, 'REDACTED', -75, 'REDACTED', 1]
[5, 'REDACTED', -77, 'REDACTED', 1]
[3, 'REDACTED', -77, 'REDACTED', 1]
[4, 'REDACTED', -78, 'REDACTED', 6]
[0, 'REDACTED', -80, 'REDACTED', 11]
[3, 'REDACTED', -80, 'REDACTED', 11]
[3, 'REDACTED', -81, 'REDACTED', 6]
[5, 'REDACTED', -82, 'REDACTED', 6]
[4, 'REDACTED', -82, 'REDACTED', 11]
[3, 'REDACTED', -82, 'REDACTED', 11]
[4, 'REDACTED', -83, 'REDACTED', 1]
[3, 'REDACTED', -84, 'REDACTED', 1]
Checking connection...
Connecting...
---> AT+CIPSTATUS
<--- b'STATUS:0\r\n\r\nOK\r\n'
Connected to None
---> AT+CWMODE?
<--- b'+CWMODE:1\r\n\r\nOK\r\n'
---> AT+CIPSTATUS
<--- b'STATUS:0\r\n\r\nOK\r\n'
---> AT+CWJAP="REDACTED","REDACTED"
<--- b'WIFI CONNECTED\r\n\r\nOK\r\n'
no IP
Failed to connect, retrying
 Didn't get IP address
---> AT+CIPSTATUS
<--- b'STATUS:5\r\n\r\nOK\r\n'
Connected to None
'''''

jerryneedell, 2019-05-29:
This is the problem - it never sees the "WIFI GOT IP" after the "WIFI CONNECTED"
I do not understand why, but this is why it is failing.

```
---> AT+CWJAP="REDACTED","REDACTED"
<--- b'WIFI CONNECTED\r\n\r\nOK\r\n'
```

jerryneedell, 2019-05-29:
In my test it shows 
```---> AT+CWJAP="Needell Airport","REDACTED"
<--- b'WIFI CONNECTED\r\nWIFI GOT IP\r\n'```

jerryneedell, 2019-05-29:
Just to be sure - I redownloaded and flashed the ESP32 Firmware on mine -- Still works fine -- same version information. I thought it was the same since I posted it ;-) but I wanted to be sure.
I have no idea what is going on. I'm trying to create a script to allow simple interaction with the AT Firmware on the Argon -- I'll post it when I have it working.

jerryneedell, 2019-05-29:
Try running this program argon_uart.py
It allow you to manually enter AT commands
In the output below I tried:
AT+GMR
AT+CWLAP
AT+CWJAP="MYSSID","PASSWORD"

Try them on yours and see what output you get
Note- there is a 3 second delay at the end of each command to make sure he transmission is complete.

```
import board
import busio
import time
import digitalio

RX = board.ESP_TX
TX = board.ESP_RX
esp_enable = digitalio.DigitalInOut(board.ESP_WIFI_EN)
esp_enable.direction = digitalio.Direction.OUTPUT
esp_cts = digitalio.DigitalInOut(board.ESP_CTS)
esp_cts.pull = digitalio.Pull.UP
esp_boot = digitalio.DigitalInOut(board.ESP_BOOT_MODE)
esp_enable.direction = digitalio.Direction.OUTPUT

esp_boot = True
time.sleep(.1)
esp_enable.value = False
time.sleep(.1)
esp_enable.value = True

with  busio.UART(TX, RX, baudrate=115200, timeout=3) as uart:
    # get first response to AT "
    uart.write(bytes("AT\r\n","utf-8"))
    getresponse = True
    while getresponse:
        response=uart.readline()
        print(response)
        if(response == b'' or response == None ):
            getresponse=False
    while True:
        print("enter AT command:")
        data=input()
        uart.write(bytes(data+"\r"+"\n","utf-8"))
        getresponse = True
        while getresponse:
            response=uart.readline()
            print(response)
            if(response == b'' or response == None ):
                getresponse=False

```

```

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.1-5-g63b253c33 on 2019-05-24; Particle Argon with nRF52840
>>> 
>>> 
>>> import argon_uart
b'ets Jun  8 2016 00:22:57\r\n'
b'\r\n'
b'rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)\r\n'
b'configsip: 0, SPIWP:0xee\r\n'
b'clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00\r\n'
b'mode:DIO, clock div:2\r\n'
b'ad:0x3fff0018,len:4\r\n'
b'd:0x3fff001c,len:6248\r\n'
b'load:0x40078000,len:10168\r\n'
b'load:0x40080400,len:6468\r\n'
b'entry 0x40080758\r\n'
b'\xfe\r\n'
b'ready\r\n'
b'WIFI CONNECTED\r\n'
b'WIFI GOT IP\r\n'
b''
enter AT command:
AT+GMR
b'AT+GMR\r\n'
b'AT version:1.3.0.0-dev(6aafec8 - Dec  7 2018 02:23:19)\r\n'
b'SDK version:v3.3-dev-343-g7fa98593b\r\n'
b'compile time:Dec 24 2018 12:27:05\r\n'
b'\r\n'
b'OK\r\n'
b''
enter AT command:
AT+CWLAP
b'AT+CWLAP\r\n'
b'+CWLAP:(4,"",-38,"7a:a7:56:ea:c5:fd",11)\r\n'
b'+CWLAP:(3,"Needell Airport",-39,"5a:a7:56:ea:c5:fd",11)\r\n'
b'+CWLAP:(3,"Needell Airport",-59,"5a:a7:56:ea:be:cd",1)\r\n'
b'+CWLAP:(4,"",-62,"7a:a7:56:ea:be:cd",1)\r\n'
b'+CWLAP:(3,"",-63,"4a:a7:56:ea:be:cd",1)\r\n'
b'+CWLAP:(3,"Needell Airport",-73,"5a:a7:56:ea:c7:0d",1)\r\n'
b'+CWLAP:(4,"",-73,"7a:a7:56:ea:c7:0d",1)\r\n'
b'+CWLAP:(3,"",-73,"4a:a7:56:ea:c7:0d",1)\r\n'
b'+CWLAP:(3,"Needell Airport",-77,"2c:7e:81:f9:8b:be",6)\r\n'
b'+CWLAP:(5,"",-79,"6e:7e:81:f9:8b:be",6)\r\n'
b'+CWLAP:(4,"",-80,"4e:7e:81:f9:8b:be",6)\r\n'
b'+CWLAP:(0,"xfinitywifi",-81,"3e:7e:81:f9:8b:be",6)\r\n'
b'+CWLAP:(3,"central2.4",-90,"00:11:32:7c:75:fc",4)\r\n'
b'+CWLAP:(3," A31A",-91,"02:11:32:7c:75:fc",4)\r\n'
b'+CWLAP:(0,"xfinitywifi",-97,"5e:be:f7:d1:3a:58",11)\r\n'
b'+CWLAP:(4,"Jquighome",-98,"54:be:f7:d1:3a:58",11)\r\n'
b'+CWLAP:(4,"",-98,"62:be:f7:d1:3a:58",11)\r\n'
b'\r\n'
b'OK\r\n'
b''
enter AT command:
AT+CWJAP="Needell Airport","REDACTED"
b'AT+CWJAP="Needell Airport","REDACTED"\r\n'
b'WIFI DISCONNECT\r\n'
b'WIFI CONNECTED\r\n'
b'WIFI GOT IP\r\n'
b'\r\n'
b'OK\r\n'
b''
enter AT command:
AT+CWJAP?
b'AT+CWJAP?\r\n'
b'+CWJAP:"Needell Airport","5a:a7:56:ea:be:cd",1,-60\r\n'
b'\r\n'
b'OK\r\n'
b''
enter AT command:
```

Arudinne, 2019-05-29:
```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.1-5-g63b253c33 on 2019-05-24; Particle Argon with nRF52840
>>> import argon_uart
b'\xffets Jun  8 2016 00:22:57\r\n'
b'\r\n'
b'rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)\r\n'
b'configsip: 0, SPIWP:0xee\r\n'
b'clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00\r\n'
b'mode:DIO, clock div:2\r\n'
b'load:0x3fff0018,len:4\r\n'
b'load:0x3fff001c,len:6248\r\n'
b'load:0x40078000,len:10168\r\n'
b'load:0x40080400,len:6468\r\n'
b'entry 0x40080758\r\n'
b'\xfe\r\n'
b'ready\r\n'
b''
enter AT command:
AT+GMR
b'WIFI CONNECTED\r\n'
b'AT+GMR\r\n'
b'AT version:1.3.0.0-dev(6aafec8 - Dec  7 2018 02:23:19)\r\n'
b'SDK version:v3.3-dev-343-g7fa98593b\r\n'
b'compile time:Dec 24 2018 12:27:05\r\n'
b'\r\n'
b'OK\r\n'
b''
enter AT command:
AT+CWLAP
b'AT+CWLAP\r\n'
b'+CWLAP:(3,"REDACTED",-79,"REDACTED",1)\r\n'
b'+CWLAP:(3,"REDACTED",-79,"REDACTED",1)\r\n'
b'+CWLAP:(5,"REDACTED",-80,"REDACTED",1)\r\n'
b'+CWLAP:(3,"REDACTED",-80,"REDACTED",6)\r\n'
b'+CWLAP:(4,"REDACTED",-82,"REDACTED",6)\r\n'
b'+CWLAP:(4,"REDACTED",-83,"REDACTED",6)\r\n'
b'+CWLAP:(3,"REDACTED",-84,"REDACTED",11)\r\n'
b'+CWLAP:(5,"REDACTED",-84,"REDACTED",11)\r\n'
b'+CWLAP:(3,"REDACTED",-85,"REDACTED",11)\r\n'
b'+CWLAP:(4,"REDACTED",-87,"REDACTED",1)\r\n'
b'+CWLAP:(3,"REDACTED",-87,"REDACTED",1)\r\n'
b'+CWLAP:(3,"REDACTED",-89,"REDACTED",6)\r\n'
b'+CWLAP:(5,"REDACTED",-93,"REDACTED",11)\r\n'
b'\r\n'
b'OK\r\n'
b''
enter AT command:
AT+CWJAP="REDACTED","REDACTED"
b'WIFI CONNECTED\r\n'
b'AT+CWJAP="REDACTED","REDACTED"\r\n'
b'WIFI CONNECTED\r\n'
b'\r\n'
b'OK\r\n'
b''
enter AT command:
AT+CWJAP?
b'AT+CWJAP?\r\n'
b'+CWJAP:"REDACTED","REDACTED",1,-77\r\n'
b'\r\n'
b'OK\r\n'
b''

===========

enter AT command:
AT+CIFSR
b'AT+CIFSR\r\n'
b'+CIFSR:STAIP,"0.0.0.0"\r\n'
b'+CIFSR:STAMAC,"30:ae:a4:d1:db:e0"\r\n'
b'\r\n'
b'OK\r\n'
b''
enter AT command:
AT+CWMODE?
b'AT+CWMODE?\r\n'
b'+CWMODE:1\r\n'
b'\r\n'
b'OK\r\n'
b''
enter AT command:
AT+CWDHCP= 1,l
b'AT+CWDHCP= 1,l\r\n'
b'ERR CODE:0x01080000\r\n'
b'\r\n'
b'ERROR\r\n'
b''

=====

enter AT command:
AT+CWJAP="REDACTED","REDACTED"
b'AT+CWJAP="REDACTED","REDACTED"\r\n'
b'WIFI CONNECTED\r\n'
b'\r\n'
b'OK\r\n'
b''
enter AT command:
AT+CIFSR
b'AT+CIFSR\r\n'
b'+CIFSR:STAIP,"0.0.0.0"\r\n'
b'+CIFSR:STAMAC,"30:ae:a4:d1:db:e0"\r\n'
b'\r\n'
b'OK\r\n'
```

I ran those commands and some additional commands and it still never gets an IP. I checked our WiFi system's management page and it does see the MAC address of the device, but it never pulls an IP. I checked the DHCP server for the guest network and it wasn't even close to being exhausted, but to be safe I went ahead and restarted the DHCP service and tried the commands to connect again but it still doesn't pull an IP.

I ran this command but the output doesn't seem to match any of the Esspressif documentation:

```
enter AT command:
AT+CWDHCP?
b'AT+CWDHCP?\r\n'
b'+CWDHCP:4\r\n'
b'OK\r\n'
```

Also, this UART output seems to indicate its set to 115201 not 115200. Maybe it's just a bug? I ran it twice and got the same output

```
enter AT command:
AT+UART?
b'AT+UART?\r\n'
b'+UART:115201,8,1,0,0\r\n'
b'\r\n'
b'OK\r\n'
b''
```

I'm not sure what other AT Commands are implemented or would provide useful output.

jerryneedell, 2019-05-29:
I get the same response to the AT+UART? command   - I don't think it is relevant.

I have no idea why it is not working for you. :-(

jerryneedell, 2019-05-29:
Is there something about your AP's that restricts devices than can get IP addresses via DHCP?

Arudinne, 2019-05-29:
Well I tried one more thing. 

I ran AT+RESTORE

at first I thought I broke it more, because it wouldn't accept the AT commands that worked previously then I thought to run

AT+CWMODE=1

Now it works:

AT+CIFSR
b'AT+CIFSR\r\n'
b'+CIFSR:STAIP,"10.10.253.38"\r\n'
b'+CIFSR:STAMAC,"30:ae:a4:d1:db:e0"\r\n'
b'\r\n'
b'OK\r\n'
b''

So I guess I'll need to run a factory reset on my other one at home.

jerryneedell, 2019-05-29:
Wow! Congratulations!  That is great to know.

Arudinne, 2019-05-29:
Looks like it's able to ping now on the Simpletest.

```
Pinging 8.8.8.8...---> AT+PING="8.8.8.8"
<--- b'+PING:24\r\n\r\nOK\r\n'
24
Pinging 8.8.8.8...---> AT+PING="8.8.8.8"
<--- b'+PING:9\r\n\r\nOK\r\n'
9
Pinging 8.8.8.8...---> AT+PING="8.8.8.8"
<--- b'+PING:11\r\n\r\nOK\r\n'
11
Pinging 8.8.8.8...---> AT+PING="8.8.8.8"
<--- b'+PING:21\r\n\r\nOK\r\n'
21
Pinging 8.8.8.8...---> AT+PING="8.8.8.8"
<--- b'+PING:32\r\n\r\nOK\r\n'
32
Pinging 8.8.8.8...---> AT+PING="8.8.8.8"
<--- b'+PING:25\r\n\r\nOK\r\n'
25
Pinging 8.8.8.8...---> AT+PING="8.8.8.8"
<--- b'+PING:10\r\n\r\nOK\r\n'
10
```

The only thing I can think of is that maybe the firmware upload didn't erase part of the flash and some the old config from when I had it on the particle cloud was stuck. Was yours ever on the particle cloud?

jerryneedell, 2019-05-29:
I'm not sure -- I thought I had tried one first, but I can be sure.  We had some early issues with the AT Firmware on some ESP32s because we were not loading everything necessary. I thought it was fixed, but perhaps there is still something missing.

Thanks for figuring this out. I'm sure it will help others.

Arudinne, 2019-05-29:
Thanks for the help!

closed 2019-05-29 by Arudinne

semajoel, 2019-10-18:
GREAT HELP, THANKS AMILLION

otherguy, 2020-12-29:
Oh my god I spent like 3 hours on this before finding this post. Thank you @Arudinne! Running this immediately fixed it!

```AT+RESTORE
AT+CWMODE=1```

Note that this was on a brand new Argon that was never connected to the particle cloud.

scarolan, 2022-06-12:
@Arudinne you are my hero.  These two commands were exactly what I needed to get this pesky Wemos D1 mini working.