adafruit/Adafruit_CircuitPython_ESP_ATcontrol #47: esp01, gc iot core and ssl

h3retek, 2021-11-03:
Hi,

Following [PyPortal IoT Plant Monitor with Google Cloud IoT Core and CircuitPython](https://learn.adafruit.com/pyportal-iot-plant-monitor-with-google-cloud-iot-core-and-circuitpython)

I'm getting

`OKError: No OK response to AT+CIPSTART="SSL","mqtt.googleapis.com",8883,10`

after

`google_mqtt.connect()`

Setup:
Pico with ESP 01

Anyone have any ideas?

I'm guesssing it has something to do with old hardware? It being an ESP8266 and all... on account of [this post](https://www.esp8266.com/viewtopic.php?p=91917).

ladyada, 2021-11-03:
yeah pretty much. this is why we stopped using ESP8266, and use ESP32 only - you need better encryption. not much we can fix with it. please try 'airlift' firmware for esp32!
https://learn.adafruit.com/adafruit-airlift-breakout

closed 2021-11-03 by ladyada

h3retek, 2021-11-04:
Ok, thanks a lot ^_^

I think, I'll just change my project a bit:

Run a ESP32 webserver and post the device status to it, then the webserver can handle Google.