adafruit/Adafruit_CircuitPython_ESP_ATcontrol #72: core API compatible rewrite of the ESP32 AT interface

bablokb, 2024-09-10:
Since this library is no longer supported and since using it is complicated, I started a rewrite that provides drop-in replacements of the relevant modules (e.g. `wifi`, `ssl` and so on) for CircuitPython boards without native wifi: https://github.com/bablokb/circuitpython-esp32at.

 The AT-APIs supported are for the ESP32C3 and ESP32C6, so old co-processors like the 8266 won't work.

Currently, all station related methods of `wifi.radio` work. UDP-clients (i.e. sending data) as well as HTTP-GET/HTTPS-GET requests work as well. Since I don't have a complete test-coverage, this does not mean it will work in all circumstances.

Feedback is of course highly welcome. If you encounter problems, just open an issue in the project repository.

ladyada, 2024-09-11:
thanks, looks like a good option for folks!

ladyada, 2024-09-11:
@TheKitty newsy